ros2/message_filters #2: Integrate changes from intel/ros2_message_filters repository

dirk-thomas, 2018-07-09:
@jwang11 @gaoethan @testkit In order to integrate the various changes from the https://github.com/intel/ros2_message_filters repository here please create separate PRs for separate /unrelated changes:

* remove no more relevant (ROS 1 specific) code
* add additional files from other repositories (e.g. `MessageEvent`) - depending on the amount of code the history might be good to keep (?)
* refactor code from `Boost` to `std`
* update source code, tests, CMake, manifest to ament / ROS 2
* (anything I missed while looking through the "diff" in the `intel` repo)

By making separate PRs with orthogonal changes it will be easier / quicker to review / iterate /merge them.

gaoethan, 2018-07-10:
okay, let me try to split them into small patches to ease this process. thanks !

testkit, 2018-07-12:
@dirk-thomas, will you pls. grant @gaoethan write access to merge the PR?

dirk-thomas, 2018-07-12:
> will you pls. grant @gaoethan write access to merge the PR?

We are happy to grant write access to you. For now I would like to focus on iterating on the PR. As commented there it needs some updates to follow the guideline we provided in this ticket.

[The idea of granting write access is not to "just merge the PR as-is". A PR should always be reviewer from another person before it is being merged.]

gaoethan, 2018-07-26:
Now because @tfoote is working on his refactoring with the branch https://github.com/ros2/message_filters/compare/refactor4 to move the message_filters here, in order to keep his coherence and make the patches conflict potentially each other, let's go ahead after @tfoote initial refactoring on a whole as an unified base for further development/issue fix. thanks !

mjcarroll, 2018-08-16:
Is this closed by ros2/message_filters#5?

mikaelarguedas, 2018-08-16:
It looks like the python tests have not been ported over yet https://github.com/ros2/message_filters/pull/5#discussion_r206355342

sloretz, 2018-08-16:
@gaoethan It looks like https://github.com/intel/ros2_message_filters/ already has the python `message_filters` code ported to ros2. May I ask you to open a PR with the python code? If I understand correctly that being merged marks when this ticked is complete.

gaoethan, 2018-08-17:
ok. no problem, let me file a PR for the python part.

gaoethan, 2018-08-19:
I have already integrated the ros2 `message_filters `python support with some previous changes of [intel/ros2_message_filters](https://github.com/intel/ros2_message_filters) and new changes to fully support of headerless messages. please find the details underneath #7 , thanks !

gaoethan, 2018-09-17:
@dirk-thomas Since C++ part had already been merged and @tfoote has already approved the python part of  #7 , so I can say that the iterating work is almost done. could you please share the write access now ? because I'll try to check to release it with bloom and fix the potential issues before releasing it to community formally.  thanks !

tfoote, 2018-09-25:
@gaoethan For the moment we'd rather stay in the loop for the release cycle for such a core package until you have a little bit more experience with maintaining packages. If you'd like to test running bloom you can go ahead and start doing that. It does not require commit access to test the process. You can create a temporary release repository for testing if you'd like to including testing the push mechanism too.

gaoethan, 2018-09-26:
@tfoote It's okay for us to run bloom with another temporary repository and thank you for you feedback.
but I think you should be more direct and specific regarding what you said for the following, especially that we're working in the open community

> For the moment we'd rather stay in the loop for the release cycle for such a core package until you have a little bit more experience with maintaining packages

I don't want to argue anything here, because it's really meaningless for any argument here, but I want to give some explanation:
- Firstly, we have to apologize that we dropped all the commit history and log unexpectedly by my another colleague while he tried to port the ROS message_filters as a new project which was hosted underneath Intel github, we really know that it's not acceptable, but it's really not done on purpose and not his essential intent, this is our badness, and thank you for your finding to give us feedback.
- Secondly, we think that all the commits should be size reasonable and easy to review, and each PR should be reviewed by the other community members and pass the backend CI/CI tests to go to the next step for merging no matter for maintainer or contributors, but you know, now the whole CI running/trigger can only be enabled by a few people (mainly in the OSRF), I really know your hard work to enable easy CI for the community, but it's really not convenient for the others and can be better in the future.
- Third, whether to maintain the project or not is not really that important for us, because what we really cares is whether our contribution can contribute values to the community, if the community can benefit from our contribution, that's our essential motivation. if not, there is nothing in it.  

Thank you for your feedback and apologize for any inconvenience !

gaoethan, 2018-10-09:
now C++ and Python interfaces have already been integrated, I think that it's time to close this ticket now.

closed 2018-10-09 by dirk-thomas