ros2/message_filters #11: Is there replace way to call connection method with signals2

yechun1, 2018-10-23:
There is one commit deleted connection method (3bfc64acdeb153f9633f59d23b4b97700e6e9548), see comment by @tfoote  that "TODO(tfoote) find why", this method will be added later? or is there any replace method could be call instead? 

the delated code:
```
Connection::Connection(const WithConnectionDisconnectFunction& func, boost::signals2::connection c)
: connection_disconnect_(func)
, connection_(c)
{
}
```

I am porting[ tf2_ros/message_filter.h](https://github.com/ros2/geometry2/blob/ros2/tf2_ros/include/tf2_ros/message_filter.h#L433) on ros2 which used this method, appreciate if anyone could help.

```
 message_filters::Connection registerFailureCallback(const FailureCallback& callback)
  {  
    std::unique_lock<std::mutex> lock(failure_signal_mutex_);
    return message_filters::Connection(std::bind(&MessageFilter::disconnectFailure, this, _1), failure_signal_.connect(callback)); 
  }
```

yechun1, 2018-10-24:
Found a thread which discussed the same issue.  
ROS2 Package Migration: Replacement for CallbackQueueInterface
https://discourse.ros.org/t/ros2-package-migration-replacement-for-callbackqueueinterface/5751

yechun1, 2018-10-26:
This issue has been discussed and answered on  [How do you use CallbackGroups as a replacement for CallbackQueues in ROS2?](https://answers.ros.org/question/300874/how-do-you-use-callbackgroups-as-a-replacement-for-callbackqueues-in-ros2/?answer=301165#post-id-301165). 

Seems currently there is no similar callback queues worked in ROS 2. So I have got trouble to port tf2_ros message_filter on ROS 2 to replace this ros::CallbackQueueInterface related code.  I'm not sure if could revert the commit [Add option to use a callback queue in the message filter](https://github.com/ros2/geometry2/commit/c7a9adbcc2c276f03d0e030e6030afbc886f65a8). Could this be a template solution   and workable for other package porting such as amcl and navigation issues?

wjwwood, 2018-10-26:
I think you will need this feature request in rclcpp: https://github.com/ros2/rclcpp/issues/519 With that you could use callback groups + executors like a callback queue from ROS 1.

gaoethan, 2018-10-30:
I think that this issue has nothing to do with ros2 message_filters, because it requires the ros2 node implementation to add API to return its callback queue (a.k.a callback groups and executors in ros2) which it uses to handle its callback. if no work around, a new PR should be redirected to rclcpp.

yechun1, 2018-12-26:
Closed this issue as  tf2_ros/message_filter.h had been ported by https://github.com/ros2/geometry2/pull/81

closed 2018-12-26 by yechun1