ros2/message_filters #20: How to use message filters in a class with Subscribers as member variables

AndreasAZiegler, 2019-04-10:
I try to get a time_synchronizer running with an Odometry and a LaserScan topic. The example provided in the README uses subscribers as normal function variables whereas in my case the subscriber are a class member variable.

In the following example (https://github.com/ros2/examples/blob/master/rclcpp/minimal_subscriber/member_function.cpp) the ```create_subscription``` method is used to initialize the subscriber class member variable. My question now is, how can I initialize a time_synchronizer? I assume I don't need the callbacks for the Odometry and the LaserScan topic but rather one callback for the synchronizer. I looked into the ```create_subscription``` method and saw that it requires a callback method so I'm wondering if I have to use something else instead of the ```create_subscription``` method?

closed 2019-04-11 by AndreasAZiegler

marting87, 2019-08-19:
For anyone that ends up here with a similar issue:

Change: 

```
timeSynchronizer_.registerCallback(
      std::bind(&PolygonExplorerNode::subscriberCallback, this,
                std::placeholders::_1, std::placeholders::_2));
```

To:

```
timeSynchronizer_.registerCallback(
     &PolygonExplorerNode::subscriberCallback, this);
```

florianspy, 2019-11-05:
Try this :
private:
	ros::NodeHandle n;
	ros::Publisher odom_pub;
	message_filters::Subscriber<Image> image_sub;
	message_filters::Subscriber<Image> image_sub2;
	TimeSynchronizer<Image, Image> sync2;
};

Harsharma2308, 2020-08-16:
The member initialisation doesn't work for me for `message_filters::Subscriber`.
This is my code- 
```
class MultiSubscriber : public rclcpp::Node
{
  public:
      
      message_filters::Subscriber<sensor_msgs::msg::Image> image_sub_;
      message_filters::Subscriber<sensor_msgs::msg::Image> depth_sub_;
      
    MultiSubscriber(const std::string& name)
    : Node(name), 
      image_sub_(message_filters::Subscriber<sensor_msgs::msg::Image>(this, "image_color_topic")),
      disparity_sub_(message_filters::Subscriber<sensor_msgs::msg::Image>(this, "image_disparity_topic"))
    {
      
      typedef message_filters::sync_policies::ApproximateTime<sensor_msgs::msg::Image, sensor_msgs::msg::Image> approximate_policy;
      message_filters::Synchronizer<approximate_policy>syncApproximate(approximate_policy(10), image_sub_, depth_sub_);

      syncApproximate.registerCallback(&MultSubscriber::disparityCb,this);
      
     
    }
    private:
    void disparityCb(const sensor_msgs::msg::Image::SharedPtr disparity_msg, const sensor_msgs::msg::Image::SharedPtr color_msg){
   //dosomething
                }
};
```

I get this error - 
```
/home/hsharma/Documents/internship/ws/ros2_ws/src/stereo_ground_surface/obstacle_detection/src/multisense_subscriber.cpp: In constructor ‘MultiSenseSubscriber::MultiSenseSubscriber(const string&)’:
/home/hsharma/Documents/internship/ws/ros2_ws/src/stereo_ground_surface/obstacle_detection/src/multisense_subscriber.cpp:68:120: error: use of deleted function ‘message_filters::Subscriber<sensor_msgs::msg::Image_<std::allocator<void> > >::Subscriber(const message_filters::Subscriber<sensor_msgs::msg::Image_<std::allocator<void> > >&)’
     : Node(name), image_sub_(message_filters::Subscriber<sensor_msgs::msg::Image>(this, "/multisense/left/image_color"))
                                                                                                                        ^
In file included from /home/hsharma/Documents/internship/ws/ros2_ws/src/stereo_ground_surface/obstacle_detection/src/multisense_subscriber.cpp:10:0:
/opt/ros/eloquent/include/message_filters/subscriber.h:102:7: note: ‘message_filters::Subscriber<sensor_msgs::msg::Image_<std::allocator<void> > >::Subscriber(const message_filters::Subscriber<sensor_msgs::msg::Image_<std::allocator<void> > >&)’ is implicitly deleted because the default definition would be ill-formed:
 class Subscriber : public SubscriberBase, public SimpleFilter<M>
       ^~~~~~~~~~
/opt/ros/eloquent/include/message_filters/subscriber.h:102:7: error: use of deleted function ‘message_filters::SimpleFilter<sensor_msgs::msg::Image_<std::allocator<void> > >::SimpleFilter(const message_filters::SimpleFilter<sensor_msgs::msg::Image_<std::allocator<void> > >&)’
In file included from /opt/ros/eloquent/include/message_filters/subscriber.h:41:0,
                 from /home/hsharma/Documents/internship/ws/ros2_ws/src/stereo_ground_surface/obstacle_detection/src/multisense_subscriber.cpp:10:
/opt/ros/eloquent/include/message_filters/simple_filter.h:57:7: note: ‘message_filters::SimpleFilter<sensor_msgs::msg::Image_<std::allocator<void> > >::SimpleFilter(const message_filters::SimpleFilter<sensor_msgs::msg::Image_<std::allocator<void> > >&)’ is implicitly deleted because the default definition would be ill-formed:
 class SimpleFilter : public noncopyable
       ^~~~~~~~~~~~
/opt/ros/eloquent/include/message_filters/simple_filter.h:57:7: error: use of deleted function ‘message_filters::noncopyable::noncopyable(const message_filters::noncopyable&)’
In file included from /opt/ros/eloquent/include/message_filters/subscriber.h:40:0,
                 from /home/hsharma/Documents/internship/ws/ros2_ws/src/stereo_ground_surface/obstacle_detection/src/multisense_subscriber.cpp:10:
/opt/ros/eloquent/include/message_filters/connection.h:51:3: note: declared here
   noncopyable( const noncopyable& ) = delete;
   ^~~~~~~~~~~
In file included from /opt/ros/eloquent/include/message_filters/subscriber.h:41:0,
                 from /home/hsharma/Documents/internship/ws/ros2_ws/src/stereo_ground_surface/obstacle_detection/src/multisense_subscriber.cpp:10:
/opt/ros/eloquent/include/message_filters/simple_filter.h:57:7: error: use of deleted function ‘message_filters::Signal1<sensor_msgs::msg::Image_<std::allocator<void> > >::Signal1(const message_filters::Signal1<sensor_msgs::msg::Image_<std::allocator<void> > >&)’
 class SimpleFilter : public noncopyable
       ^~~~~~~~~~~~
In file included from /opt/ros/eloquent/include/message_filters/simple_filter.h:43:0,
                 from /opt/ros/eloquent/include/message_filters/subscriber.h:41,
                 from /home/hsharma/Documents/internship/ws/ros2_ws/src/stereo_ground_surface/obstacle_detection/src/multisense_subscriber.cpp:10:
/opt/ros/eloquent/include/message_filters/signal1.h:82:7: note: ‘message_filters::Signal1<sensor_msgs::msg::Image_<std::allocator<void> > >::Signal1(const message_filters::Signal1<sensor_msgs::msg::Image_<std::allocator<void> > >&)’ is implicitly deleted because the default definition would be ill-formed:
 class Signal1
       ^~~~~~~
/opt/ros/eloquent/include/message_filters/signal1.h:82:7: error: use of deleted function ‘std::mutex::mutex(const std::mutex&)’
In file included from /usr/include/c++/7/mutex:43:0,
                 from /usr/include/c++/7/future:38,
                 from /opt/ros/eloquent/include/rclcpp/executors.hpp:18,
                 from /opt/ros/eloquent/include/rclcpp/rclcpp.hpp:145,
                 from /home/hsharma/Documents/internship/ws/ros2_ws/src/stereo_ground_surface/obstacle_detection/src/multisense_subscriber.cpp:2:
/usr/include/c++/7/bits/std_mutex.h:97:5: note: declared here
     mutex(const mutex&) = delete;

```

clalancette, 2020-08-17:
> The member initialisation doesn't work for me for `message_filters::Subscriber`.

I'll suggest you open up a new issue, otherwise this will probably get lost.

Harsharma2308, 2020-08-18:
If someone gets stuck on this,  someone answered  my question on ROS answers  - https://answers.ros.org/question/359614/ros2-message-filters-in-a-class-initializing-message_filterssubscriber-object/