ros2/message_filters #21: TimeSequencer.simple test hanging in CI

pbaughman, 2019-04-11:
One of our CI jobs just failed because the TimeSequencer.simple unit test hung:

```
      Start  4: message_filters-test_time_sequencer

4: Test command: /usr/bin/python3 "-u" "opt_path/share/ament_cmake_test/cmake/run_test.py" "ws_path/build/message_filters/test_results/message_filters/message_filters-test_time_sequencer.gtest.xml" "--package-name" "message_filters" "--output-file" "ws_path/build/message_filters/ament_cmake_gtest/message_filters-test_time_sequencer.txt" "--command" "ws_path/build/message_filters/message_filters-test_time_sequencer" "--gtest_output=xml:ws_path/build/message_filters/test_results/message_filters/message_filters-test_time_sequencer.gtest.xml"
4: Test timeout computed to be: 60
4: -- run_test.py: invoking following command in 'ws_path/src/path/message_filters':
4:  - ws_path/build/message_filters/message_filters-test_time_sequencer --gtest_output=xml:ws_path/build/message_filters/test_results/message_filters/message_filters-test_time_sequencer.gtest.xml
4: [==========] Running 3 tests from 1 test case.
4: [----------] Global test environment set-up.
4: [----------] 3 tests from TimeSequencer
4: [ RUN      ] TimeSequencer.simple
 4/13 Test  #4: message_filters-test_time_sequencer ............***Timeout  60.01 sec
test 5
      Start  5: message_filters-test_subscriber
```

A passing run in our CI takes about half a second to run, so this isn't a case of the test just taking a little longer than normal.

It's almost certainly getting stuck in `rclcpp::spin_some(node)` on [line 97](https://github.com/ros2/message_filters/blob/master/test/time_sequencer_unittest.cpp#L97) or a few lines later.  The queue of things to handle is either building up faster than spin_some can process, or some work item in the queue is adding a work item to the queue so spin_some never finishes.

I need to dive into this a little more to see what's being tested, but maybe we could spin until something specific happens, or spin until future complete?  If someone reads this and has a suggestion, please chime in

pbaughman, 2019-04-11:
It looks like we've got `message_filters.message_filters-test_fuzz` marked as problematic too, possibly for the same reason

ZhenshengLee, 2024-01-12:
happens in rolling, output in in below.

```sh
zs@zs-vm-2204:~/ga_ws/ga_ros/colcon/build/message_filters$ ./message_filters-test_fuzz 
[==========] Running 3 tests from 3 test suites.
[----------] Global test environment set-up.
[----------] 1 test from TimeSequencer
[ RUN      ] TimeSequencer.fuzz_sequencer
unknown file: Failure
C++ exception with description "can't compare times with different time sources" thrown in the test body.
[  FAILED  ] TimeSequencer.fuzz_sequencer (26 ms)
[----------] 1 test from TimeSequencer (26 ms total)

[----------] 1 test from TimeSynchronizer
[ RUN      ] TimeSynchronizer.fuzz_synchronizer
[       OK ] TimeSynchronizer.fuzz_synchronizer (5013 ms)
[----------] 1 test from TimeSynchronizer (5013 ms total)

[----------] 1 test from Subscriber
[ RUN      ] Subscriber.fuzz_subscriber
[       OK ] Subscriber.fuzz_subscriber (5022 ms)
[----------] 1 test from Subscriber (5022 ms total)

[----------] Global test environment tear-down
[==========] 3 tests from 3 test suites ran. (10061 ms total)
[  PASSED  ] 2 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TimeSequencer.fuzz_sequencer

 1 FAILED TEST
```