ros2/message_filters #22: Signal9.h "Reference "_1" is ambigious"

klintan, 2019-04-29:
```template<typename P0, typename P1>
  Connection addCallback(void(*callback)(P0, P1))
  {
    using namespace std::placeholders;
    return addCallback(std::function<void(P0, P1, NullP, NullP, NullP, NullP, NullP, NullP, NullP)>(std::bind(callback, _1, _2)));
  }

  template<typename P0, typename P1, typename P2>
  Connection addCallback(void(*callback)(P0, P1, P2))
  {
    using namespace std::placeholders;
    return addCallback(std::function<void(P0, P1, P2, NullP, NullP, NullP, NullP, NullP, NullP)>(std::bind(callback, _1, _2, _3)));
  }```

I use Boost in the parent project, including the message filters and get the "Reference "_1" is ambigious" everywhere where the placeholders is defined. Even though the `using namespace std::placeholders;` is defined in each function scope. 

It should lookup the variable in the local namespace first I presume? as stated here https://stackoverflow.com/a/9381283/1829511 but it seems that the global namespace somehow takes precedence? Could this happen if somewhere any of my libs has a `using namespace boost` for instance? 
Any other ideas why this happens ? 

My solution was to define the fully qualified name  ```std::placeholders::_1``` but that is a bit of a pain.

dirk-thomas, 2019-04-29:
Are you invoking `using namespace`...` anywhere globally in a header in your code? If yes, than that is a bad idea and you might want to fix this in your code.

klintan, 2019-04-29:
I'll have an extra look in all headers and see if I can find any.

closed 2020-04-28 by klintan