ros2/message_filters #34: TypeError: unhashable type: 'Time' error

jdddog, 2019-07-09:
In the TimeSynchronizer class the following error is thrown because msg.header.stamp ([rcl_interfaces/builtin_interfaces/msg/Time.msg](https://github.com/ros2/rcl_interfaces/blob/master/builtin_interfaces/msg/Time.msg)) is not hashable (perhaps it was in ROS 1), see below:

```bash
  File "/home/jamie/my_ros2_ws/install/message_filters/lib/python3.6/site-packages/message_filters/__init__.py", line 227, in add
    my_queue[msg.header.stamp] = msg
TypeError: unhashable type: 'Time'
```
This occurs at [message_filters/\_\_init\_\_.py#L227](https://github.com/ros2/message_filters/blob/master/src/message_filters/__init__.py#L227).

sloretz, 2019-10-03:
I see the same error on Ubuntu Bionic in Dashing. A workaround is to use `ApproximateTimeSyncrhonizer` with a small `slop` value.

dirk-thomas, 2019-10-24:
A message class is in general not hashable (beyond using the object id). The rational is simple: if you would like all fields to be part of a the hash you need to enforce that the field values don't change after an object has been hashed since otherwise the hash would change.

closed 2019-11-09 by sloretz

charlielito, 2020-11-24:
Same error here, I don't think this issue should be closed, since it is a bug and it does not work when you try to use the python version of `TimeSyncrhonizer`. Additionally a README would be appreciated :)

charlielito, 2020-12-04:
Nevermind, just find out this was fixed in `master` but I am using the dashing version that has a lot of bugs :/

PR #33 (2019-11-09): Fix TypeError: unhashable type: 'Time' error
Fixes #34 

In the TimeSynchronizer class the following error is thrown because msg.header.stamp (builtin_interfaces/Time) is not hashable (perhaps it was in ROS 1), see below:

```bash
  File "/home/jamie/my_ros2_ws/install/message_filters/lib/python3.6/site-packages/message_filters/__init__.py", line 227, in add
    my_queue[msg.header.stamp] = msg
TypeError: unhashable type: 'Time'
```
This occurs at [message_filters/\_\_init\_\_.py#L227](https://github.com/ros2/message_filters/blob/master/src/message_filters/__init__.py#L227).

I've just taken t [... cut]