ros2/message_filters #36: callback is not executed with message filters

chaoli2, 2019-08-19:
I failed to have callback function called using message filters. So I created a minimal project to verify the interface and functionality of message filters. In

 this project, I add two publishers to publish identical messages into two topics. Then in main function, I try to synchronize them with message filters. But the callback functions still has no response. What's wrong with my [code](https://github.com/chaoli2/minimal_test_project)? I ask for your kindly help. Thanks very much.

OS: Ubuntu 18.04
ROS2: Dashing release

publisher.cpp
```
#include <rclcpp/rclcpp.hpp>
#include <rclcpp/time.hpp>                                                                            
#include <message_filters/subscriber.h>                                                               
#include <message_filters/time_synchronizer.h>
#include <std_msgs/msg/header.hpp>
                                                                                                      
#include <iostream>                                                                                   
#include <memory>                                                                                     
#include <unistd.h>                                                                                   
                                                                                                      
class Publisher1 : public rclcpp::Node                                                                
{                                                                                                     
public:                                                                                               
  Publisher1() : Node("publisher") {};
};                                                                                                    
                                                                                                      

int main(int argc, char ** argv)                                                                      
{                                                                                                     
  rclcpp::init(argc, argv);                                                                           
  auto node = rclcpp::Node::make_shared("publisher");
                                                                                                      
  rclcpp::Publisher<std_msgs::msg::Header>::SharedPtr pub1, pub2;
  pub1 = node->create_publisher<std_msgs::msg::Header>("/publish1", 16);
  pub2 = node->create_publisher<std_msgs::msg::Header>("/publish2", 16);
                                                                                                      
  std_msgs::msg::Header msg1, msg2;                                                                   
                                                                                                      
  while (rclcpp::ok())                                                                                
  {                                                                                                   
    std::cout << "publishing" << std::endl;                                                           
    rclcpp::Time time= node->now();                                                                   
    msg1.stamp = time;                                                                                
    msg2.stamp = time;                                                                                
    pub1->publish(msg1);
    pub2->publish(msg2);                                                                              
    rclcpp::spin_some(node);
    sleep(1);
  }                                                                                                   
}
```

main.cpp
```
#include <rclcpp/rclcpp.hpp>
#include <message_filters/subscriber.h>
#include <message_filters/time_synchronizer.h>
#include <std_msgs/msg/header.hpp>

#include <iostream>
#include <memory>

class MessageFilterTest : public rclcpp::Node
{
public:
  MessageFilterTest() : Node("test")
  {
    rclcpp::Node::SharedPtr node = std::shared_ptr<rclcpp::Node>(this);
    
    cam_sub_ = std::make_unique<camSub>(node, "/publisher1");
    obj_sub_ = std::make_unique<objSub>(node, "/publisher2");
    sync_sub_ = std::make_unique<sync>(*cam_sub_, *obj_sub_, 10);
    sync_sub_->registerCallback(std::bind(&MessageFilterTest::callback, this, std::placeholders::_1, std::placeholders::_2));
  };

private:
  using camSub = message_filters::Subscriber<std_msgs::msg::Header>;
  using objSub = message_filters::Subscriber<std_msgs::msg::Header>;

  using sync =
     message_filters::TimeSynchronizer<std_msgs::msg::Header, std_msgs::msg::Header>;

  std::unique_ptr<camSub> cam_sub_;
  std::unique_ptr<objSub> obj_sub_;
  std::unique_ptr<sync> sync_sub_;

  void callback(const std_msgs::msg::Header::ConstSharedPtr msg1, const std_msgs::msg::Header::ConstSharedPtr msg2) { std::cout << "callback" << std::endl; };
};

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);
  //rclcpp::executors::SingleThreadedExecutor exec;
  auto message_filter_test_node = std::make_shared<MessageFilterTest>();

  rclcpp::spin(message_filter_test_node);
  rclcpp::shutdown();

  return 0;
}
```

AmAppu, 2020-03-31:
hi,
did you get any solution for the problem?
I have the same problem.
Callback works while I set registecallback() from main(), but no output when set from a class member function.

hidmic, 2020-04-16:
@chaoli2 I see a local `rclcpp::Node` being instantiated within an `rclcpp::Node` constructor and then dropped at the end of the scope. But I'd think you've realized about that in the code you refer to. I'll close this issue now. 

In the future, you're better off asking this sort of questions in [answers.ros.org](https://answers.ros.org/).

closed 2020-04-16 by hidmic