ros2/message_filters #39: message filter doesn't allow non-default quality of service

Jconn, 2019-10-08:
The subscription created in the Subscriber __init__ has a hard-coded queue depth of 10. If `qos_profile` is set as a named param like I am doing:

```
        self.pc_filter = message_filters.Subscriber(
                self,
                PointCloud2,
                constants.PC2_TOPIC,
                qos_profile = qos_profile_sensor_data
                )

```
Then an exception is thrown, because qos_profile is being set twice.
I am throwing around a lot of pointcloud data, and I really want to be able to specify a best effort quality of service For some of my filters. 

Thank you for the clear type annotations, making this easy to track down.

for reference, the problem __init__ function is shown below:
```
class Subscriber(SimpleFilter):
    def __init__(self, *args, **kwargs):
        SimpleFilter.__init__(self)
        self.node = args[0]
        self.topic = args[2]
        self.sub = self.node.create_subscription(*args[1:], self.callback, 10, **kwargs)
```

wjwwood, 2019-11-14:
@claireyywang can you close this or update it as needed?

claireyywang, 2019-11-14:
> @claireyywang can you close this or update it as needed?

Yeah sure my bad. Closing as the PR is merged.

closed 2019-11-14 by claireyywang