ros2/message_filters #49: Potential memory leaks in message_filters

SteveMacenski, 2020-10-29:
Hi,

Per https://github.com/ros-planning/navigation2/issues/1889, we're seeing memory leaks in Navigation2 in servers that make use of message filters that we believe message filters is the root cause (potentially could be TF2 ROS within message filters).

clalancette, 2020-10-29:
It would be interesting to retry the test with Rolling.  We've fixed a few memory leaks in tf2_ros there, but we haven't been able to backport all of them to Foxy (since they break ABI; see https://github.com/ros2/geometry2/pull/330 for an in-progress PR).

SteveMacenski, 2020-10-29:
Would the master branch be acceptable as well? We don't use rolling, we run master.

clalancette, 2020-10-29:
> Would the master branch be acceptable as well? We don't use rolling, we run master nightly builds.

Yeah, that would also work.  It's preferable, honestly, since sometimes there is a delay between when we merge things and when it makes it to the Rolling packages.

SteveMacenski, 2020-10-29:
I just don't use rolling on anything I maintain because I refuse to deal with the onslaught of emails it generates. I actually had to drop maintaining image pipeline for that reason (some small things break, suddenly I have `3 * number of packages in repo` emails in my inbox, daily).

clalancette, 2020-11-18:
I believe this should be fixed in Foxy and on the master branches now (via https://github.com/ros2/geometry2/pull/330).  @SteveMacenski can you confirm?

SteveMacenski, 2020-11-18:
Yes, we closed it in regards to that patch, thanks for bringing that home

closed 2020-11-18 by SteveMacenski