ros2/message_filters #52: Can't get the name of the publisher

giuseppelamarca, 2021-02-02:
HI I'm tryng to get the name of the publisher but I noticed that there is no an equivalent method as in ROS1 called getPublisherName(). Is it missing or there is another way to get the name of the publisher? Thanks for your time

sloretz, 2021-02-04:
It doesn't seem like this is an issue related to `message_filters`. Please kindly ask these sorts of questions on https://answers.ros.org per the guidelines in https://www.ros.org/support/

closed 2021-02-04 by sloretz

Federico-Ciuffardi, 2021-08-30:
Sorry to insist, but I'm not convinced this is not related to `message_filters`. 

On the line 72 of [message_filtess/message_event.h](https://github.com/ros2/message_filters/blob/foxy/include/message_filters/message_event.h) is the following comment:

> Useful if you need to retrieve meta-data about the message, such as the full connection header, or the publisher's node name

But I did not found any way of retrieve the  full connection header or the publisher's node name. 

As @giuseppelamarca says there was a function [getPublisher()](http://docs.ros.org/en/noetic/api/roscpp_traits/html/message__event_8h_source.html#l00176) in ROS1 implementation of `message_event` and also, as the previous comment indicates there was another function to retrieve the full connection header [getConnectionHeader()](http://docs.ros.org/en/noetic/api/roscpp_traits/html/message__event_8h_source.html#l00170).

Why doesn't the current `message_event` in ROS2 include these functionalities? Is it because there is another way to recover this data in ROS2?

muttistefano, 2022-05-29:
HI, 
did you manage to find a solution to this ? 
Thanks

Federico-Ciuffardi, 2022-05-30:
No :(