ros2/message_filters #55: Current message_filters do not reflex tutorial

Sorawit-DeepX, 2021-05-17:
I'm new to ROS and ROS2, so I follow the tutorial and find some disturbing problem of inconsistence between 2 version.
I want to leave this message because this problem take me 3 hours to solve.
ROS tutorial
```
import message_filters
from sensor_msgs.msg import Image, CameraInfo

def callback(image, camera_info):
  # Solve all of perception here...

image_sub = message_filters.Subscriber('image', Image)
info_sub = message_filters.Subscriber('camera_info', CameraInfo)

ts = message_filters.TimeSynchronizer([image_sub, info_sub], 10)
ts.registerCallback(callback)
rospy.spin()
```
This indicate message_filters.Subscriber(topic, data_type)

but in ROS2 is
```
    def __init__(self, *args, **kwargs):
        SimpleFilter.__init__(self)
        self.node = args[0]
        self.topic = args[2]
        kwargs.setdefault('qos_profile', 10)
        self.sub = self.node.create_subscription(*args[1:], self.callback, **kwargs)
```
Which is message_filters.Subscriber(node, ???, topic)

Should we make it more consistent?

clalancette, 2021-05-17:
Can you please give a pointer to which tutorial you were following?

Hi-Zed, 2021-11-03:
I guess he is referencing this page:
http://wiki.ros.org/message_filters

I am also having some issues with using message filters on Python. I can't find an up to date example for ROS2

clalancette, 2022-02-03:
OK, the message filters tutorial in ROS 1 isn't going to work as-is for ROS 2.  As such, this is essentially a request for a new tutorial for ROS 2, which is very reasonable.

I'll go ahead and move this over to the ros2_documentation repository, which is where we track those things.