ros2/message_filters #61: TimeSynchronizer while subscribing to two camera topics using ROS2 - Fox

niranjanravilevelup, 2021-08-13:
Hi,

 I am working with ROS 2- foxy distribution. I have two USB cameras simultaneously running and publishing images in topics - /usb_cam_0/image_raw and /usb_cam_1/image_raw . I want to subscribe the topics and concatenate the images and save them in a folder.

I went through this https://answers.ros.org/question/8112... and tried to adapt the same logic as,

subscription_1 = message_filters.Subscriber('/usb_cam_0/image_raw', Image)

subscription_2 = message_filters.Subscriber('/usb_cam_1/image_raw', Image)

message_filters.ApproximateTimeSynchronizer([subscription_1, subscription_2],10,1)

But this script throws me an error in the subscriber class of message filter. The error was

File "/home/device/messageFilter/src/install/message_filters/lib/python3.6/site-packages/message_filters/__init__.py", line 78, in __init__ self.topic = args[2] IndexError: tuple index out of range

Unable to clear this error. Other approach which worked was

subscription_1 = self.create_subscription(Image, '/usb_cam_0/image_raw', imageClass.camera_0_callback, 1)

But here i would have to use two separate callback functions which I don't want to. I cloned the message filters package from this repository but still the same error persists. 

The first logic works was from the dicsussion with ROS. But does it have a big difference with ROS2 - foxy? I am new to this platform. Any guidance would be helpful.

Regards
Niranjan

fhoerbinger, 2021-08-25:
Hi! 
I think the ROS2 message_filter.Subscriber expects the node as the first argument. So something like this should work:
subscription_1 = message_filters.Subscriber(self, Image, '/usb_cam_0/image_raw')

clalancette, 2022-02-03:
> I think the ROS2 message_filter.Subscriber expects the node as the first argument. So something like this should work:
> subscription_1 = message_filters.Subscriber(self, Image, '/usb_cam_0/image_raw')

Yes, that's exactly right.

Since this is a question, I'm going to close this out.  In the future, please ask questions like this on https://answers.ros.org, which is our central question and answer site.

closed 2022-02-03 by clalancette