ros2/message_filters #69: If someone has a compilation problem like /opt/ros/foxy/include/message_filters/message_traits.h:89:30: error: ‘class rclcpp::Time’ has no member named ‘sec’ , I will provide my own solution

IDCARDBORING, 2022-01-12:
/opt/ros/foxy/include/message_filters/message_traits.h:89:30: error: ‘class rclcpp::Time’ has no member named ‘sec’
   89 |     return rclcpp::Time(stam.sec, stam.nanosec);
      |                         ~~~~~^~~
/opt/ros/foxy/include/message_filters/message_traits.h:89:40: error: ‘class rclcpp::Time’ has no member named ‘nanosec’; did you mean ‘nanoseconds’?
   89 |     return rclcpp::Time(stam.sec, stam.nanosec);

You can find ”message_traits.h“ in your local folder  ，for example /opt/ros/foxy/include/message_filters/message_traits.h 
and
85line
replace
struct TimeStamp<M, typename std::enable_if<HasHeader<M>::value>::type >
{
  static rclcpp::Time value(const M& m) {
    auto stamp = m.header.stamp;
    auto sec=stamp.seconds();
    auto nanosec=stamp.nanoseconds();
    return rclcpp::Time(sec, nanosec);
  }
};

clalancette, 2022-01-12:
This looks like you are probably trying to compile the latest master branch of this repository against Foxy, which will not work.  Instead, you need to checkout the `foxy` branch to compile this against Foxy.

In the future, please ask these kinds of questions on https://answers.ros.org, which is our central question and answers site.  You'll get a better answer there.  Closing this for now.

closed 2022-01-12 by clalancette

IDCARDBORING, 2022-01-12:
> This looks like you are probably trying to compile the latest master branch of this repository against Foxy, which will not work. Instead, you need to checkout the `foxy` branch to compile this [...]

hey！ I'd like to know if you want information fusion between two built-in interfaces（like sensor_msg/image and geomory_msg/pose）. Is there an example？