ros2/message_filters #74: C++ equivalent of `allow_headerless` synchronization

akgoins, 2022-06-06:
I know that you can set `allow_headerless = True` in python, but is there an equivalent option for C++?  Are there plans to add it in the future?

clalancette, 2022-06-07:
It looks like there is nothing equivalent in C++ right now.  Certainly we would be happy to review a pull request that added it to the C++ side of things, so please consider contributing.

EsipovPA, 2025-07-02:
Hello everyone! 
I've opened the [PR with my take on this issue](https://github.com/ros2/message_filters/pull/195). It would be great if anyone could review it.

I'm not sure if the approach I've decided to use correlates with the design philosophy of this specific project, so I'm open to design improvement suggestions.

EsipovPA, 2025-07-07:
🚀