ros2/message_filters #91: [Question] Why the limit of synchronizer is 9 topics?

Tumb1eweed, 2023-04-07:
Since c++11 already support variadic template

clalancette, 2023-04-07:
> Since c++11 already support variadic template, should synchronizer have no topic number limit or is it another way to achieve this without modifying it?

I think it is mostly the fact that the synchronizer was written before C++11.  We'd be fine with having it rewritten with variadic templates, just nobody has had time to do it.  If you are interested in looking into it and opening a PR, we'd be happy to review.

Tumb1eweed, 2023-04-11:
> I think it is mostly the fact that the synchronizer was written before C++11. We'd be fine with having it rewritten with variadic templates, just nobody has had time to do it. If you are interested [...]

Oh, i see the PR now, thanks for answering!

closed 2023-04-11 by Tumb1eweed