ros2/message_filters #97: Support for custom timestamp

LastStarDust, 2023-05-22:
Hello!
Thank you for developing such a useful package. It has become an irreplaceable piece of our system.

In our system, the timestamp is expressed as a single uint64 number with nanosecond precision. Even if converting from `nanosec` to `sec` + `nanosec` is perfectly feasible, it would be cleaner if we could just use the timestamp as is.

Hence my question: would it be possible to use a custom definition of [std_msgs/msg/Header Message](https://docs.ros2.org/foxy/api/std_msgs/index-msg.html) for time synchronization with `TimeSynchronizer` or `ApproximateTimeSynchronizer`?

Concretely, I would like to use a header whose timestamp is a single `uint64` stamp with nanosecond precision, instead of two `int32` and `uint32` stamps with second and nanosecond precision respectively, as in the default ROS2 definition.

If this feature is not implemented yet, I would like to contribute with a pull request. Just let me know if there is any request or suggestion on your side.

Regards
Giorgio

LastStarDust, 2023-05-22:
@clalancette Sorry for not being clearer.

I am referring to the `TimeSynchronizer` class, which uses the message timestamp contained in the header to synchronize multiple ROS messages.

Please refer to [here](https://github.com/ros2/message_filters/blob/rolling/include/message_filters/time_synchronizer.h#L50) and [here](https://github.com/ros2/message_filters/blob/rolling/test/test_synchronizer.cpp#L52),

clalancette, 2023-08-03:
Sorry, now I see what you mean.

I think that as it stands, that won't really work because the code expects those fields to be a particular type.  I think you'd have to reimplement parts of https://github.com/ros2/message_filters/blob/rolling/include/message_filters/message_traits.h to make it work properly.

LastStarDust, 2023-08-04:
@clalancette 
Thank you for the hint.
Assuming I make it work, how gladly would you accept a pull request about this feature?

clalancette, 2023-08-04:
> Assuming I make it work, how gladly would you accept a pull request about this feature?

Please feel free to open a PR if you implement something.  As long as it doesn't break backwards compatibility (i.e. the default is still to use `std_msgs/msg/Header`), I think we can probably take it.

LastStarDust, 2023-08-04:
I do not plan on working on it any time soon, but I am happy to hear that you are open to including this feature.
I might open a PR someday. Until then feel free to close this issue if you prefer.

lix19937, 2024-08-31:
I also need this feature.

lix19937, 2024-08-31:
It seems user override the  `TimeStamp` will works.      

```cpp  
namespace message_filters
{
namespace message_traits
{
template<>
struct TimeStamp<Msg>
{
  static rclcpp::Time value(const Msg & m)
  {
    return m.user_stamp;
  }
};
}  // namespace message_traits
}  // namespace message_filters
```

closed 2026-08-21 by ahcorde