ros2/message_filters #110: Is there a bug in Latest time policy rates_ initialization?

wuchenhaogit, 2024-01-21:
In include/message_filters/sync_policies/latest_time.h

rates_ is array of struct Rate. I thought that its index corresponds to channel index. But its initialization in function initialize_rate<i>() doesn't put the struct (including alphas and prev) to index "i". Instead "push_back" is used. As a result,  the index of rates_ seems to represent the order of arrival of the first message of each channel. This appears confusing to me. How can  #I properly set the alphas for each channel?

Here is the source code of initialize_rate():
  template<int i>
  void initialize_rate()
  {
    if (rate_configs_.size() > 0U) {
      double rate_ema_alpha{Rate::DEFAULT_RATE_EMA_ALPHA};
      double error_ema_alpha{Rate::DEFAULT_ERROR_EMA_ALPHA};
      double rate_step_change_margin_factor{Rate::DEFAULT_MARGIN_FACTOR};
      if (rate_configs_.size() == RealTypeCount::value) {
        std::tie (
          rate_ema_alpha,
          error_ema_alpha,
          rate_step_change_margin_factor) = rate_configs_[i];
      } else if (rate_configs_.size() == 1U) {
        std::tie (
          rate_ema_alpha,
          error_ema_alpha,
          rate_step_change_margin_factor) = rate_configs_[0U];
      }
      rates_.push_back(
        Rate(
          ros_clock_->now(),
          rate_ema_alpha,
          error_ema_alpha,
          rate_step_change_margin_factor));
    } else {
      rates_.push_back(Rate(ros_clock_->now()));
    }
  }

andermi, 2024-01-23:
Confirmed.  I'll submit a PR. Thanks for finding this!

EsipovPA, 2026-11-16:
There were no activity in this issue for the last 2 years. I guess I could take this one. Could anyone assign it to me please?