ros2/message_filters #112: [ADD] Node interfaces support

wodtko, 2024-02-12:
At work, we have been using message filters for a while now.
While transitioning to `LifecycleNodes`, we noticed that the interface of the message subscriber still relies on a Node type, which supports `create_subscription` with a specific signature.
The intended way of the `rclcpp::NodeInterfaces` seems to circumvent the use of any Node type and use specific interfaces instead.

We are currently working on implementing similar changes in other projects (see [image_transport](https://github.com/authaldo/image_common/pull/1/files) and [image_transport_plugins](https://github.com/authaldo/image_transport_plugins/pull/1/files)).
This is why we depend on similar changes here.

Are there any reasons against such a change?
Since we (especially @authaldo) are working on it anyway, we would be happy to contribute a PR.

mjcarroll, 2024-02-12:
No reasons against the change, likely just a lack of someone contributing the feature.  Please open a PR and we would be happy to iterate on it.

wodtko, 2024-02-12:
awesome, thanks for the fast reply, @authaldo created a respective PR #113

PR #113 (2025-03-27): More generic subscriber implementation using NodeInterfaces from rclcpp
Fixes #112 

To be more precise, these changes allow using message_filters::Subscriber in situations, where no full rclcpp::Node / rclcpp_lifecycle::LifecycleNode is available by requiring only the relevant interfaces.
closed 2025-03-27 by ahcorde