ros2/message_filters #130: Extend Tutorials Section

CursedRock17, 2024-06-08:
As per the title we could keep expanding the [tutorials section](https://github.com/ros2/message_filters/tree/rolling/doc/Tutorials) of our `message_filters` documentation because you can never have enough documentation. Specifically https://github.com/ros2/message_filters/pull/119#discussion_r1604565909 specifies for us to add an [ApproximateEpsilonTime](https://github.com/ros2/message_filters/blob/rolling/include/message_filters/sync_policies/approximate_epsilon_time.h) tutorials. That being said the following policies and objects could technically have their own tutorials:

- [ ] [ApproximateEpsilonTime](https://github.com/ros2/message_filters/blob/rolling/include/message_filters/sync_policies/approximate_epsilon_time.h) (C++)
- [ ] [LatestTime](https://github.com/ros2/message_filters/blob/ac26507686ea8533504bb917128659951029759d/include/message_filters/sync_policies/latest_time.h) (C++)
- [ ] [Chain](https://github.com/ros2/message_filters/blob/ac26507686ea8533504bb917128659951029759d/include/message_filters/chain.h#L104) (C++)
- [ ] [Cache](https://github.com/ros2/message_filters/blob/ac26507686ea8533504bb917128659951029759d/include/message_filters/cache.h#L62) (C++)
- [ ] [Cache](https://github.com/ros2/message_filters/blob/ac26507686ea8533504bb917128659951029759d/src/message_filters/__init__.py#L93) (Python)

Possibly:

- [ ] [Custom Simple Filter](https://github.com/ros2/message_filters/blob/ac26507686ea8533504bb917128659951029759d/include/message_filters/simple_filter.h#L51) (C++ - Intermediate)
- [ ] [Custom Simple Filter](https://github.com/ros2/message_filters/blob/ac26507686ea8533504bb917128659951029759d/src/message_filters/__init__.py#L46) (Python - Intermediate)

I should be able to get to most of these relatively fast, but they should be in one location ready for tracking. Additionally since most of tfhe documentation is hosted on `message_filters` this shouldn't need to go to `ros2_documentation`

fujitatomoya, 2024-06-27:
/label `help-wanted` @ahcorde can you add the label, cz i do not have power to do that.

EsipovPA, 2025-07-01:
I thik this issue is not closed yet. I've added only one tutorial and working on them one by one. Is it possible to reopen this issue? @CursedRock17

CursedRock17, 2025-07-01:
@EsipovPA i would agree there's still lots to do, I did mark off the Cache filter in the tasks list though.

EsipovPA, 2025-07-16:
Added PR with the tutorial for `C++` for `Cache`. Reviews are welcome! `:)`

closed 2025-09-22 by ahcorde

EsipovPA, 2025-09-22:
I think this issue is not completely closed. Maybe reopen it again? And add a checkmark for Chain (C++)?

EsipovPA, 2025-11-23:
It took slightly more time that I've expected to do the `Custom SimpleFilter` tutorials. I've completed the tutorial for `Python`. I'll open a `PR` during the course of this week. I need to read it out for errors, typos and stuff like that. Sorry for the delay.

EsipovPA, 2025-11-29:
Opened a PR with SimpleFilter inheritance tutorial. Will be greatfull for reviews! `:)`

EsipovPA, 2026-02-17:
Started working on tutorial for `ApproximateEpsilonTime`. Should not there be also a python implementation? I may add one, if it sounds reasonable, but later.

PR #185 (2025-06-27): Add tutorial for Cache filter for Python
## Description

Added a tutorial for Cache message filter class for Python programming launguage.

Partially fixes #130 

### Did you use Generative AI?

### Additional Information
PR #187 (2025-06-27): Add tutorial for Cache filter for Python (backport #185)
## Description

Added a tutorial for Cache message filter class for Python programming launguage.

Partially fixes #130 

### Did you use Generative AI?

### Additional Information
<hr>This is an automatic backport of pull request #185 done by [Mergify](https://mergify.com).
PR #188 (2025-06-27): Add tutorial for Cache filter for Python (backport #185)
## Description

Added a tutorial for Cache message filter class for Python programming launguage.

Partially fixes #130 

### Did you use Generative AI?

### Additional Information
<hr>This is an automatic backport of pull request #185 done by [Mergify](https://mergify.com).
PR #189 (2025-06-27): Add tutorial for Cache filter for Python (backport #185)
## Description

Added a tutorial for Cache message filter class for Python programming launguage.

Partially fixes #130 

### Did you use Generative AI?

### Additional Information
<hr>This is an automatic backport of pull request #185 done by [Mergify](https://mergify.com).