ros2/message_filters #160: Jazzy branch not updated

JorgePerC, 2024-12-06:
Hi! I was trying to build nav2 on ROS Jazzy, and downloaded this repo as a submodule in my workspace using the jazzy branch. My build was failing...
It was missing the _message_filters/subscriber.hpp_ file. Which is present as a .h, I realized the rolling branch actually has it.

Just letting you know. 

If I can help in any way, let me know

clalancette, 2024-12-06:
> It was missing the _message_filters/subscriber.hpp_ file. Which is present as a .h, I realized the rolling branch actually has it.

Yes, that is basically on purpose.  We only migrated to using `.hpp` files *after* Jazzy, so Jazzy and before should still be using the `.h` files.

JorgePerC, 2024-12-10:
You're right. I didn't specify the branch of the sub-module, and it was creating dependency issues.
Thanks a lot

closed 2024-12-10 by JorgePerC