ros2/message_filters #162: Build Failure for ROS 2 jazzy: fatal error: message_filters/synchronizer.hpp: No such file or directory

theksg, 2024-12-30:
ROS 2 Version: Jazzy
Ubuntu(WSL): 24.04.01 LTS

I am trying to do "colcon build" but I am getting this error:
colcon build
```
Starting >>> msg_filter_demo
--- stderr: msg_filter_demo
/home/sakshamg/msg_filter_ws2/src/msg_filter_demo/src/approximate_time_synchronizer.cpp:7:10: fatal error: message_filters/synchronizer.hpp: No such file or directory
    7 | #include "message_filters/synchronizer.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/approximate_time_sync.dir/build.make:76: CMakeFiles/approximate_time_sync.dir/src/approximate_time_synchronizer.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/approximate_time_sync.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< msg_filter_demo [1.37s, exited with code 2]

Summary: 0 packages finished [1.88s]
  1 package failed: msg_filter_demo
  1 package had stderr output: msg_filter_demo
```
  
 Any file under message_filters folder are not able to detect. 
 I have manually verified the presence of message_filter package and have also seen the presence of files I am using in the code.
 
 My code files are as following:

approximate_time_synchronizer.cpp

```
#include "rclcpp/rclcpp.hpp"

#include <chrono>
#include <functional>
#include <memory>

#include "sensor_msgs/msg/temperature.hpp"
#include "sensor_msgs/msg/fluid_pressure.hpp"

#include "message_filters/synchronizer.hpp"
#include "message_filters/sync_policies/approximate_time.hpp"
#include "message_filters/subscriber.hpp"

using namespace std::chrono_literals;

using std::placeholders::_1;
using std::placeholders::_2;

class TimeSyncNode : public rclcpp::Node
{
  public:
    TimeSyncNode() : Node("sync_node")
    {
    rclcpp::QoS qos = rclcpp::QoS(10);
    temp_pub = this->create_publisher<sensor_msgs::msg::Temperature>("temp", qos);
    fluid_pub = this->create_publisher<sensor_msgs::msg::FluidPressure>("fluid", qos);

    temp_sub.subscribe(this, "temp", qos);
    fluid_sub.subscribe(this, "fluid", qos);

    timer = this->create_wall_timer(500ms, std::bind(&TimeSyncNode::TimerCallback, this));
    second_timer = this->create_wall_timer(550ms, std::bind(&TimeSyncNode::SecondTimerCallback, this));

    uint32_t queue_size = 10;
    sync = std::make_shared<message_filters::Synchronizer<message_filters::sync_policies::
        ApproximateTime<sensor_msgs::msg::Temperature, sensor_msgs::msg::FluidPressure>>>(
        message_filters::sync_policies::ApproximateTime<sensor_msgs::msg::Temperature,
        sensor_msgs::msg::FluidPressure>(queue_size), temp_sub, fluid_sub);

    sync->setAgePenalty(0.50);
    sync->registerCallback(std::bind(&TimeSyncNode::SyncCallback, this, _1, _2));

    }

  private:
    rclcpp::Publisher<sensor_msgs::msg::Temperature>::SharedPtr temp_pub;
    rclcpp::Publisher<sensor_msgs::msg::FluidPressure>::SharedPtr fluid_pub;
    message_filters::Subscriber<sensor_msgs::msg::Temperature> temp_sub;
    message_filters::Subscriber<sensor_msgs::msg::FluidPressure> fluid_sub;
    std::shared_ptr<message_filters::Synchronizer<message_filters::sync_policies::ApproximateTime<
      sensor_msgs::msg::Temperature, sensor_msgs::msg::FluidPressure>>> sync;
    rclcpp::TimerBase::SharedPtr timer;
    rclcpp::TimerBase::SharedPtr second_timer;

    void SyncCallback(const sensor_msgs::msg::Temperature::ConstSharedPtr & temp,
        const sensor_msgs::msg::FluidPressure::ConstSharedPtr & fluid)
    {
    RCLCPP_INFO(this->get_logger(), "Sync callback with %u and %u as times",
        temp->header.stamp.sec, fluid->header.stamp.sec);
    if (temp->temperature > 2.0)
    {
        sensor_msgs::msg::FluidPressure new_fluid;
        new_fluid.header.stamp = rclcpp::Clock().now();
        new_fluid.header.frame_id = "test";
        new_fluid.fluid_pressure = 2.5;
        fluid_pub->publish(new_fluid);
    }
    }

    void TimerCallback()
    {
    sensor_msgs::msg::Temperature temp;
    auto now = this->get_clock()->now();
    temp.header.stamp = now;
    temp.header.frame_id = "test";
    temp.temperature = 1.0;
    temp_pub->publish(temp);
    }

    void SecondTimerCallback()
    {
    sensor_msgs::msg::FluidPressure fluid;
    auto now = this->get_clock()->now();
    fluid.header.stamp = now;
    fluid.header.frame_id = "test";
    fluid.fluid_pressure = 2.0;
    fluid_pub->publish(fluid);
    }
};

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);
  auto node = std::make_shared<TimeSyncNode>();
  rclcpp::spin(node);
  rclcpp::shutdown();

  return 0;
}
```
CMakeLists.txt

```
cmake_minimum_required(VERSION 3.8)
project(msg_filter_demo)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
# uncomment the following section in order to fill in
# further dependencies manually.
find_package(rclcpp REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(message_filters REQUIRED)

add_executable(approximate_time_sync src/approximate_time_synchronizer.cpp)
ament_target_dependencies(approximate_time_sync rclcpp sensor_msgs message_filters)

install(TARGETS
    approximate_time_sync
    DESTINATION lib/${PROJECT_NAME})

ament_package()
```
 package.xml

```
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>msg_filter_demo</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="sakshamg@todo.todo">sakshamg</maintainer>
  <license>Apache-2.0</license>

  <buildtool_depend>ament_cmake</buildtool_depend>

  <test_depend>ament_lint_auto</test_depend>
  <test_depend>ament_lint_common</test_depend>

  <export>
    <build_type>ament_cmake</build_type>
    <depend>rclcpp</depend>
    <depend>sensor_msgs</depend>
    <depend>message_filters</depend>
  </export>
</package>
```

clalancette, 2024-12-30:
You'll need to checkout the `jazzy` branch to compile this package against Jazzy.

closed 2024-12-30 by clalancette

theksg, 2024-12-30:
> You'll need to checkout the `jazzy` branch to compile this package against Jazzy.

I have the jazzy branch only

theksg, 2024-12-30:
oh, got it, I was using code from the example on rolling branch, i changed the code with jazzy branch one, thanks