ros2/message_filters #169: ApproximateTimeSynchronizer callback doesn't execute

belalhmedan90, 2025-03-13:
The ApproximateTimeSynchronizer callback is never called due to deadlock maybe, here is a minimal example:

```py
#!/usr/bin/env python
# coding: utf-8

import rclpy
from rclpy.node import Node
from rclpy.executors import MultiThreadedExecutor
from rclpy.callback_groups import ReentrantCallbackGroup, MutuallyExclusiveCallbackGroup

from message_filters import Subscriber, ApproximateTimeSynchronizer
from std_msgs.msg import Bool
from sensor_msgs.msg import Image
from std_srvs.srv import Empty

from time import time

class ATS(Node):
    def __init__(self):
        super().__init__(node_name='ats_node')
        self._active = False
        
        self.create_subscription(Bool, '/ctr_ats', callback=self._activate, qos_profile=1, callback_group=MutuallyExclusiveCallbackGroup())
        self._color_sub = Subscriber(self, Image, "/camera/camera/color/image_rect_color")
        self._depth_sub = Subscriber(self, Image, "/camera/camera/aligned_depth_to_color/image_raw")
        self._ts = ApproximateTimeSynchronizer(
            [self._color_sub, self._depth_sub], 1, 0.1
        )
        self._ts.registerCallback(self._ats_image_cb)

    def _activate(self, msg: Bool):
        self._active = msg.data

    def _ats_image_cb(self):
        self._logger.warning(message="Outside ATS callback")
        if self._active:
            self._logger.warning(message="Inside ATS callback")

class SrvServer(Node):
    def __init__(self):
        super().__init__(node_name='ats_node')
        self._cam_active_pub = self.create_publisher(Bool, '/ctr_ats', qos_profile=1, callback_group=ReentrantCallbackGroup())
        self._srv = self.create_service(Empty, '/do_something', callback=self._do_anything , callback_group=ReentrantCallbackGroup())

    def _do_anything(self, request, response):
        self._logger.info(message="Inside do_anything service!")

        ctrl_msg = Bool()
        ctrl_msg.data = True
        self._cam_active_pub.publish(ctrl_msg)

        start_time = time()
        while (time() - start_time) < 5:
            rclpy.spin_once(self, timeout_sec=0.1)
            self._logger.info("Spinning inside srv cb")

        ctrl_msg.data = False
        self._cam_active_pub.publish(ctrl_msg)
        
        return response

def main(args=None):
    rclpy.init(args=args)
    try:
        srv_node = SrvServer()
        sub_node = ATS()
        executor = MultiThreadedExecutor()
        executor.add_node(srv_node)
        executor.add_node(sub_node)
        try:
            executor.spin()
        finally:
            executor.shutdown()
    finally:
        rclpy.shutdown()

if __name__ == '__main__':
    main()
```

> P.S: you also can use the [minimal package here](https://github.com/belalhmedan90/ros2_test_ats)

christophebedard, 2025-03-13:
Similar issue: https://github.com/ros2/rclpy/issues/1432

belalhmedan90, 2025-03-13:
turns that callback function signature is wrong `_ats_image_cb`

closed 2025-03-13 by belalhmedan90