ros2/message_filters #175: TimeSequencer using system clock instead of node clock

ichatzinikolaidis, 2025-04-11:
Hi,

I noticed that TimeSequencer is using the system clock (both [here](https://github.com/ros2/message_filters/blob/e58552d2dfe6bb26eed2504aedc0b32a29b4a310/include/message_filters/time_sequencer.hpp#L202) and in the definition of `last_time_`) instead of the node's clock. Could this be problematic when running a simulation? Shouldn't it use the node's clock, which can be configured to use either sim time or system time by setting the `use_sim_time` parameter?

MichaelOrlov, 2025-04-24:
It seems to support sim_time need to use clock from the underlying node.  
However, @mjcarroll needs to clarify.

EsipovPA, 2026-05-04:
@ahcorde Hello. Found this issue. Seems like the related PR is already merged. Should not this issue be closed? If I am not mistaken, the [PR](https://github.com/ros2/message_filters/pull/202) fixes the issue.

closed 2026-05-05 by ahcorde