ros2/message_filters #200: Inconsistency between C++ and Python implementation for ExactTimeSynchronizer

authaldo, 2025-08-05:
### Operating System:

```not relevant```

### ROS version or commit hash:

jazzy

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

The bug has been discovered while trying to include a C++ node with an exact time synchronizer policy in the [deterministic execution framework (ros2_def)](https://github.com/uulm-mrm/ros2_def). In short, the framework replicates the synchronizer policy in python in order to determine after which input and output of the node has to be expected. 

Reproducing the issue should however be possible with a simpler setup: 
Take two input streams and cause a reordering of samples. More specifically, having the two tuples (x1, x2) at timestamp t1 and (y1, y2) at timestamp t2 (with t1 < t2), the filter should for instance receive the inputs in the following order: x1, y1, y2 (causing a signal), x2. 
Depending on the chosen implementation the receival of x2 causes another callback since x1 is kept.

### Expected behavior

Given the same sequence of message inputs, the C++ and Python implementation should lead to the same sequence of synchronized outputs.

### Actual behavior

The C++ implementation drops all messages which are older than the latest signal time, while the python implementation retains them.

### Additional information

I discovered this already a while ago but it seems like I never opened an issue. By chance, I came across a similar issue in another repository (https://github.com/ros2/geometry2/issues/818), reminding me of this.
While skimming the updated code it seems like the implementation didn't change that much. Hence, I am still confident that the issue persists. However, I haven't explicitly retested it with the current rolling branch.
In case this issue has already been fixed in the past please close this issue.

More details:
The [C++ implementation](https://github.com/ros2/message_filters/blob/rolling/include/message_filters/sync_policies/exact_time.hpp#L153) erases all inputs older than the last signal time.

In contrast, the [python implementation](https://github.com/ros2/message_filters/blob/77c31e69acbb21e221fc521c1b652bf0c4d1cb45/src/message_filters/__init__.py#L233) keeps these older samples, thus, potentially causing another signalling once the missing message of the older tuple is received.

Out of curiosity:
What's the motivation behind the two independent implementations (in contrast to Python bindings)? As a side note: the two implementations do not even follow the same naming pattern for the policies.

For my use case I fixed it using the [following commit](https://github.com/authaldo/message_filters/tree/feat_more_cpp_python_consistency). If that is a suitable fix, I am happy to rebase it and contribute a PR.

MichaelOrlov, 2025-08-14:
Hi @authaldo. We considered this issue at the weekly ROS 2 triage meeting. We agree on the conclusion about the bug report and proposed bugfix.  
Please follow up and create a PR with the proposed fix. Please also add some unit tests to cover changes.

EsipovPA, 2025-10-31:
Hello @authaldo! Is it Ok if I take your suggested commit and convert it into PR? If you do not have the time for that? Not trying to be offensive. just trying to be helpful 😃

authaldo, 2025-10-31:
Hi @EsipovPA,

please feel free to cherry-pick the commit and convert it to an PR! It would be really helpful, as I won't find the time for that in the near future!

EsipovPA, 2025-11-23:
Status update: Failed to get some time to work on this issue. Started trying to reproduce on rolling today. Work in progress.

EsipovPA, 2025-12-17:
Status update:

Manged to carve out some time to work on issues here. I've managed to reproduce the bug with this test, added to the `directed.py` (Think of a better file naming maybe? #186 ) test file:

```
    def test_time_synchronizer_old_msgs_drop(self):
        def collector_callback(msg1, msg2, collector):
            collector.append((msg1, msg2))

        collector = []

        filter_0 = MockFilter()
        filter_1 = MockFilter()
        ts = TimeSynchronizer([filter_0, filter_1], 10)
        ts.registerCallback(
            functools.partial(
                collector_callback,
                collector=collector,
            )
        )

        t1 = 0
        t2 = 1

        x0 = MockMessage(t1, 1)
        x1 = MockMessage(t1, 2)

        y0 = MockMessage(t2, 1)
        y1 = MockMessage(t2, 2)

        filter_0.signalMessage(x0)
        assert len(collector) == 0

        filter_1.signalMessage(y1)
        assert len(collector) == 0

        filter_0.signalMessage(y0)
        assert len(collector) == 1
        assert collector[0] == (y0, y1)

        filter_1.signalMessage(x1)
        assert len(collector) == 1
        assert collector[0] == (y0, y1)
```
Will add it to the PR. Not sure if I should add the same test for CPP. I guess it is better to do so. Working on a proposed fix. 

As a sidenote, is this really an expected behavior? Should this kind of filter remove all the messages with timestamps older than latest? Just to be sure. @MichaelOrlov